96boards/mezzanine-community #1: Add a license to the repository

mwelling, 2017-07-24:
Here are licenses that I have used:
https://creativecommons.org/licenses/by-sa/4.0/
http://www.ohwr.org/attachments/2388/cern_ohl_v_1_2.txt

Here is a more LGPL like license for hardware if you think it might be better:
https://creativecommons.org/licenses/by/4.0/

The template released by Linaro is licensed using a BSD license:
https://github.com/96boards/96boards-kicad-mezzanine-template/blob/master/COPYING

If we try to pull in this template we need to consider that.

mwelling, 2017-07-26:
Another license to consider:
http://solderpad.org/licenses/

The template would need to be released by Linaro if we switch from the BSD license.

sdrobertw, 2017-07-26:
Michael would you mind summarizing anything I will need to bring past Linaro with regard to bringing template over to this repository?

My understanding is that the template is under BSD, and if we decide to move from BSD, we will need Linaro's permission to do this. Is this correct?

If this is the case, we will need to decide on the license before I bring it up to anyone within Linaro. This way, they will know what it is being transfer to.

Thanks!

mwelling, 2017-07-26:
You would just need permission from them to change the license if we see it fit.

Of the open hardware licenses these are more permissive:
https://creativecommons.org/licenses/by/4.0/
http://solderpad.org/licenses/

If they want to keep the license as is, I would probably think it best to follow the same for the rest of the repository.

sdrobertw, 2017-07-26:
@mwelling I personally like the creative commons license. Let me try to address this by EoW, and come back with an answer to see if it is okay to change license.

Thanks!

sdrobertw, 2017-08-10:
Are we satisfied with the licenses that are in place?

I think all we are missing is the "Suggested / recommended" license procedure when porting a template and turning it into an actual board.

Let's keep this open until that doc surfaces.