96boards/mezzanine-community #5: I2S Mezzanine

mwelling, 2017-07-24:
Consider making an I2S audio mezzanine to work with all 96boards CE spec boards.

Mani-Sadhasivam, 2017-08-23:
Nice to have:

1. Mainline supported 1.8v compatible audio codec
2. Use of on board oscillators to avoid getting MCLK from CE board
3. Provision to connect external I2S inputs
4. Exposing LS header (probably 2.54mm)

niruyadla, 2017-12-06:
We are looking for  I2S audio mezzanine to work with Hikey960 board.

It is nice to have 4 DMICs (or 4 PDM microphones) input and stereo output.

mwelling, 2017-12-07:
@niruyadla Is there a single I2S codec that can do this?

niruyadla, 2017-12-07:
As per this document https://github.com/96boards/documentation/blob/master/ConsumerEdition/HiKey960/HardwareDocs/HiKey960_SoC_Reference_Manual.pdf

Part number Hi6403 seems to  be supporting MIC x 4

mwelling, 2017-12-07:
Okay I did a bit of investigation. The Hi6403 is not available from the typical vendors and has a bit of HiSilicon tie-in giving them a bit of an advantage over base boards for other SoC vendors. Also I am not seeing mainline support for this codec.

If HiSilicon supports the development maybe I would consider it but otherwise I would try to find a codec that is more SoC agnostic and hopefully supported in mainline.

Mani-Sadhasivam, 2017-12-07:
I agree with @mwelling here, codec should be board agnostic! Also, having 4 DMIC inputs seems to be an interesting feature to have. This will extend the application capabilities of the board to Voice acquisition etc... We could eventually use this board to develop something like Echo dot.

RajPawate, 2017-12-09:
Hello Mani, and Michael Welling,
I work at Cadence and we are working on offloading tasks from App processor to HiFi 3 DSP which is embedded in the SoC of Hikey960.  Happy to see that you are interested in working on a multi-mic data acquisition card for Hikey960.  We also wish to do the same.    You might be interested in looking at http://www.ti.com/tool/TIDA-01454#technicaldocuments
This is dual PCM1864 circular microphone board reference design.  
How to connect this to Hikey960?
We need a minimum of 4 mic in (ideally 8 mic) as input.  And as Niranjan and Mani says we could do a Echo dot like product demo.
Another requirement is to have the microphone data go directly to HiFi 3 DSP via I2S.  Would you be able to investigate how to connect the TI board to the I2S of the Hikey960?
There are two issues here: (1) How to connect TI board to I2S of Hikey960 board electrically? (2) How to write DSP software on HiFi 3 DSP to access these 8 microphones (or at least 4 of them) with minimal latency?

sowjanya-tumu, 2018-06-04:
@sdrobertw , do we have any update on this ?
I see HiSilicon merged some code changes on kernel to support I2S.
@Mani-Sadhasivam Do you know when can we get mezzanine audio codec board for Hikey960?

niruyadla, 2018-06-04:
@sdrobertw , do we have any update on this ?
I see HiSilicon merged some code changes on kernel to support I2S for Hikey960 
@sdrobertw @Mani-Sadhasivam, Do you know when can we get mezzanine audio codec board for Hikey960?

sdrobertw, 2018-06-04:
@mwelling 

@Mani-Sadhasivam @niruyadla and @RajPawate are the ones who will need to provide update.

mwelling, 2018-06-04:
I have been holding off on the design waiting for more information from @Mani-Sadhasivam. There was talk of another company contributing the design which is more than welcome. Hopefully we will have some more information by the bi-weekly meeting next week.

Mani-Sadhasivam, 2018-06-05:
@niruyadla @mwelling I have prepared some recommendations for the I2S Mezzanine which I will share it in our next Mezzanine meeting! Please let me know if we can discuss it here itself before the meeting.

niruyadla, 2018-06-05:
@Mani-Sadhasivam ,Please let me know when is next Mezzanine meeting?

Mani-Sadhasivam, 2018-06-05:
@niruyadla Monday, June 11 @ 9.30 to 10 PM (Indian Standard Time)

Mani-Sadhasivam, 2018-06-08:
@mwelling @niruyadla @sdrobertw @ric96 

Here is the sample draft for the I2S Mezzanine:

https://docs.google.com/document/d/1dAwuQvU_V4PVBUMsx15CKj5A87g79PSl6L5iz3neU_U/edit?usp=sharing

I will populate this until our next meeting, please let me know your thoughts.

Mani-Sadhasivam, 2018-06-11:
@niruyadla We have logged the today's meeting notes in #64. It would be great if you let us know if you guys can support this Mezzanine by any means!

Mani-Sadhasivam, 2018-11-15:
@mwelling @niruyadla @ric96 

I have asked my college junior to come up with a low cost I2S Mezzanine design and he is up with the schematic and BoM. Please help with the review.

https://drive.google.com/file/d/1Gv76kbddEiU5T8BnT2cOk3aLxDisOm2Q/view?usp=sharing
https://drive.google.com/file/d/1fzCWMdBwlzLDHY_UTjSRuHBHb54GQ8ME/view?usp=sharing

Note: This is not intended to compete with Adam's Mezzanine but I wanted this to be a cheaper one.

Mani-Sadhasivam, 2018-11-15:
Codec used is SGTL5000 and is well supported in mainline linux kernel.

Mani-Sadhasivam, 2018-11-22:
Design reviewed by @mwelling offline. Will share the updated design shortly.