96boards/mezzanine-community #10: Add high speed connector to mezzanine template.

mwelling, 2017-08-06:
The mezzanine template currently only has low speed connector options.

sdrobertw, 2017-08-07:
Let's start a to-do list on the template README.md

I have already modified the section in PR

mwelling, 2017-08-07:
Okay, I merged just merged your PR. Go ahead and add it to the README.

GustavoRP, 2017-08-08:
The Altium template already has the high-speed connector. However, I would like someone to check if it is in the exactly right position because the mechanical documentation I found did not have enough information about this connector.

mwelling, 2017-08-08:
I will take a look before I merge it.

sdrobertw, 2017-08-10:
This template has been merged, can we close this issue?

GustavoRP, 2017-08-10:
The high-speed connector still needs to be added to the KiCad project. Unfortunately, I have no experience with it, but here is the footprint specification if someone wants to add it on the board:
http://portal.fciconnect.com/Comergent/fci/drawing/61083.pdf

mwelling, 2017-08-10:
I will take care of it.

GustavoRP, 2017-08-11:
we can close the issue, then.

mwelling, 2017-08-11:
After I take care of it. :)

mwelling, 2017-08-11:
KiCAD template has been updated to include the high-speed connector.

closed 2017-08-11 by mwelling