96boards/mezzanine-community #31: Thermal mitigation with mezzanine fan

dlezcano, 2017-10-13:
We noticed some boards can be working out of the thermal envelope and some others have a thermal behavior with fast/high amplitude.

Some of us did use a big fan powered by an USB relay which is not the best solution to do active cooling on the boards (but better than nothing).

It could be interesting to create a mezzanine board with slow speed connector deport in order to still add the console and in the center of the board, a 40x40 fan driven by a PWM processor via I2C communication with the board.

Why using a PWM processor ? We could use a gpio + pwm duty cycles to drive the fan via a mosfet but that introduces at the kernel level an non negligible activity to generate the PWM cycles and that will strongly interact with the power management.

The interesting points are:
 - most of the software pieces (perhaps all) are already in the kernel.
 - cheap components
 - easy implementation

In addition, a couple of temperature sensors could be added next to the fan on the below face.

There are a bunchs of I2C pwm controller in the Linux kernel directory drivers/hwmon. It is very important to choose the right controller (eg. being able to choose 0-255 steps for duty cycles would be very nice - some controller are short for that).

This fan could be driven by the thermal governor to increase/decrease the speed or set a fixed fan speed to emulate a specific thermal envelope (eg. phone).

I believe this project could be interesting for the server side or any 96boards with thermal mitigation needs.

sdrobertw, 2017-10-13:
@mwelling 

What do you think of this one? Seems interesting!

mwelling, 2017-10-13:
Interesting. Thermal considerations seem to be overlooked on most of the CE boards and on the spec.
The majority of the CE boards (besides DB410C) expect passive cooling from the top side I believe.
Using a fan might work but limits the applications to main carrier functions because the mezzanine slot will be consumed by the fan.

dlezcano, 2017-10-16:
I thought we can keep the slow connector on the fan board as we will be using DC_IN, 5v, and i2c, no ?

dlezcano, 2017-10-16:
- TC655 for PWM controller
 - BS170 for PWM control of the fan
 - a 40x40x10 12v fan 100mA max, 1.2W.

The fan is powered via DC_IN and driven by the BS170. The PWM controller is powered by the 5V and communicate via i2c.
Won't it work if we keep the 40pins slow connector on the fan board allowing us to stack a mezzanine board or console?

glikely, 2017-10-17:
The UART board is due for a redesign. It would be worth plopping down the PWM circuit and a connector for the fan while keeping the form factor tiny. That would allow the fan to be positioned as needed. For instance, if a mezzanine is being used, then the fan would need to be mounted vertically at the side of the board to draw air across the heatsink.

It would be a separate design to create brackets for various fan positions.

ric96, 2017-10-19:
we can use a side blowing fan like the ones used in laptops
https://www.geekbuying.com/item/Cooling-FAN-FOR-Khadas-Vim-Development-Board-376317.html