96boards/mezzanine-community #39: Meeting Agenda 04th Dec 2017

ric96, 2017-12-04:
# Agenda
- Fritzing template: got corrected pin pitch from the fritzing forums
- @mwelling: Updates on GPS Mezzanine?
- @sdrobertw: updates on Akira's Drone Mezzanine?

Tagging:
@mwelling 
@glikely 
@sdrobertw 
@GustavoRP 
@rafaelchrist

sdrobertw, 2017-12-04:
@ric96 

No updates on Akira's drone board just yet. We hope to have this board added to repo asap.

mwelling, 2017-12-04:
Send the first GPS mezzanine PCB out to OSHPark for fab.

ric96, 2017-12-12:
# Meeting Notes
- Expecting Akira's Drone Mezzanine to be pushed into the repo soon
- People are interested in using this repo to host other 96boards related hardware designs, but it's decided to keep the scope of this repo limited to mezzanine for now

closed 2019-11-25 by ric96