96boards/mezzanine-community #45: Meeting: Agenda - 22, Jan 2018

sdrobertw, 2018-01-08:
**Meeting time:** 5:00PM UTC

Please comment for next meeting's agenda items:

@glikely 
@mwelling 
@ric96 
@sdrobertw
@krisik70

Notes will be added to this card after the meeting.

ric96, 2018-01-22:
The expected due date for e-ink mezzanine has been extended to allow more time for testing and board revs.

sdrobertw, 2018-01-22:
Chat:

Ragnar.
R
Hi, I'm early. Time to grab some tea...
I still need to make coffee
https://github.com/96boards/mezzanine-community/issues/45
Ragnar.
R
nag, nag... https://github.com/96boards/mezzanine-community still states wrong time 
lol
Ragnar.
R
cute
Guest 1
G1
Good evening people
Ragnar.
R
Mike, did you add ferrite bead to the GPS. Also no 100 nF cap at the gpsmodule?
Guest 1
G1
@Robert: It's my first time in the mezzanine meetup. Listening in for now
@Pan: Thanks
Grant (ARM)
G(
PCA9306

closed 2019-11-25 by ric96