96boards/mezzanine-community #50: Meeting: Agenda - 05, March 2018

ric96, 2018-02-26:
Meeting time: 5:00PM UTC

Please comment for next meeting's agenda items:

@glikely
@mwelling
@ric96
@sdrobertw
@krisik70

Notes will be added to this card after the meeting.

sdrobertw, 2018-03-05:
- Micro Mezzanine approval has cleared (AFAIK). We will be expecting some contribution from Barry soon.
- Still waiting on Katagarsu Red Mezzanine contribution from Akira
- Migrating UART Mezzanine over to Repo soon
- Work on e-ink mezzanine to come after Connect HKG18

ric96, 2018-03-05:
- Chris Tyler is fine with us hosting the 2mm to 2.54 mm header mezzanine, although he would like a newer revision to be added.

ukBaz, 2018-03-05:
Does this mean the meeting has happened?

The meeting is scheduled for 17:00 UTC. 
https://github.com/96boards/mezzanine-community#meeting-notes
Currently it is 16:45 UTC

ric96, 2018-03-05:
@ukBaz
it's going to happen in about 15min, these are the topics we will be discussing in greater detail

sdrobertw, 2018-03-05:
https://upverter.com/ukBaz/0f2a430ab12afa2e/micromez/

sdrobertw, 2018-03-05:
@ukBaz Will need to update that time. Thanks for letting us know!

ric96, 2018-03-05:
@sdrobertw time is fine

sdrobertw, 2018-03-05:
- Dmitry: Mezzanine that brings in a battery and touch screen. 
- Ragnar: This kicad tutorial series was recommended on a local forum recently: https://www.youtube.com/watch?v=JN_Y93RTdSo&list=PL5iUxv3Op2fOpVASHvcpM2O4UO0yMfKJi
- Barry: https://www.buydisplay.com/default/10-1-inch-1024x600-50-pins-tft-lcd-display-w-touch-panel-and-touch-driver

sdrobertw, 2018-03-05:
http://www.guru3d.com/articles-pages/wireless-graphics-card-kfa2-gtx-460-wdhi-review,1.html

sdrobertw, 2018-03-05:
Attendees:

- Barry
- Dmitry
- Ivan Farkas
- Michael Welling
- Pan
- Ragnar
- Sahaj
- Robert Wolff

sdrobertw, 2018-03-05:
KiCAD Discussion as repo standard: http://kicad-pcb.org/

closed 2019-11-25 by ric96