96boards/mezzanine-community #51: Question: how much space above USB ports

Genoil, 2018-03-08:
I've just started working on a mezzanine based on your template (thanks!). Although I understand the meaning of the keepout areas, I was wondering how much room there really is between the USB type A ports and the mated mezzazine. With a PCB distance of 7mm and a USB-A receptacle of ±5.5mm, I should be able to place some small 1mm height parts there, right? Or is this asking for problems?

(btw I am using both 60pin and 40pin headers)

mwelling, 2018-03-08:
I would not recommend any through-hole or bottom side components in these keep out areas.

Genoil, 2018-03-08:
It is only an FPC header for connecting a DSI panel but I think I found a better place for it already. Thanks.

closed 2018-03-08 by Genoil