96boards/mezzanine-community #52: Eagle naming of differentials pair signals

Genoil, 2018-03-09:
For Autodesk Eagle to support differential pair routing, it is required that the signal names have the suffixes _P (for positive) and _N (for negative) instead of + and -. It would make sense to update the Eagle template per these requirements.

mwelling, 2018-03-10:
If you can do a pull request, we will accept this update.

Genoil, 2018-03-11:
On 2nd thought I'm not entirely sure this is beneficial. The routing improvements made in Eagle since the Autodesk acquisition (walkaround, push 'n shove, etc.) don't seem to work for differential pairs (yet). Until then, it seems better to route the pairs individually and there don't have the suggested renaming.

closed 2018-03-11 by Genoil