96boards/mezzanine-community #58: Meeting: Agenda 16th April 2018

ric96, 2018-04-30:
Meeting time: 5:00PM UTC

Please comment for next meeting's agenda items:

@glikely
@mwelling
@ric96
@sdrobertw
@krisik70

sdrobertw, 2018-04-30:
- Michael Welling not able to join
- Mani: Open issue, Cadence expressed need for I2S mezzanine. Suggestion is to not make a applications specific mezzanine. Cadence want it for Audio application stuff. Mani suggests a codec supported by mainline Linux. Since we do not have I2S mezzanine yet, testing audio out is hard to test.
   - Offer task to Frontline electronics. Hoping they will make the mezzanine. 
   - https://github.com/96boards/mezzanine-community/issues/5
- Resources/News section for repo

sdrobertw, 2018-04-30:
Chat Log:

Sahaj
S
just a couple more minutes
https://youtu.be/oVjZrUncDj4
Mani
M
https://github.com/96boards/mezzanine-community/issues/5
Dmitry
D
We still have issues to get from HiSi support for I2S0 (from Low Speed IO)
https://discuss.96boards.org/t/device-tree-for-i2s0-on-hikey-960/3911
We had in the past Lemaker LCD (no tough though)
http://www.lenovator.com/product/102.html
No, I am fine

closed 2019-11-25 by ric96