96boards/mezzanine-community #64: Meeting: Agenda 11th Jun 2018

ric96, 2018-06-11:
Meeting time: 4:00PM UTC
Please comment for next meeting's agenda items:
@glikely
@mwelling
@ric96
@sdrobertw
@krisik70
@Mani-Sadhasivam

Mani-Sadhasivam, 2018-06-11:
I2S Mezzanine

- There will be no commitment from the company I referred earlier
- It will be purely based on @mwelling and community
- Requirement spec: https://tinyurl.com/y7s4s4ok

ASerbinski, 2018-07-06:
Hi, has anyone actually begun working on this board? The reason I'm asking is that I'm currently working on a mezzanine board that covers most of the requirements of this one, except the headphone out. The audio aspects of my board include the following;

ADC: pcm1865
DAC: 2x pcm5142

I have it set up with a 1x stereo microphone input, 2x stereo line in, 1x stereo AMFM radio, 2x stereo line OUT. Both of the LS-I2S lines (input and output) are attached to the pcm1865, the pcm1865 i2s-out2 is split to both of the pcm5142 (they're arranged in a dual-stereo configuration). Audio output from the SBC can be mixed with any selection of analog sources in the pcm1865 before being forwarded on to the DACs, and any mix of analog sources can be fed back to the SBC.

In addition to the 3 codecs, my board also has the following;

AMP: STPA003
AMFM: TEF6686
RTC: PCF85063A
MCU: ATSAMD21E18A-A
2x high side switches for 5v and 12v controls
3x optocouplers for input signals

Now obviously my board is a lot more automotive focused, and physically larger (78x123 mm) than I'm sure your objective is with this, but sharing some of the features, at least as far as audio goes, could be mutually beneficial. For instance, my (working) Alsa machine driver; https://gitlab.com/HiKey960-Car/android_kernel_linaro_hikey/commit/03378e6c52395b40ce4deb28fc70e985ed78a9c7

And when I say that I'm working on it, I mean that I have a prototype printed, assembled (by hand), and fully functional.

mwelling, 2018-07-06:
Might want to brand it more specifically to the intended use case.

We are happy to host outside contributions if they are released as open source hardware.

Automotive Mezzanine?

Remember we are trying to provide blocks for people to get so if you have the features required of the mezzanine then we don't need another reference with the same IP.

ASerbinski, 2018-07-06:
I haven't addressed branding yet.

I don't know if I quite understand what you are suggesting. That my board be offered instead of i2s mezzanine? Or that is2 mezzanine be based directly on my schematics?

If the first, I don't think that my board would be a suitable choice for someone who is looking to add sound to something that maybe sits on their desk. To start with, my BOM alone is a lot more than I would expect anyone to want to pay for just a sound card, and on top of that, the car interface is quite different than what would be needed and expected. Mine really is purpose built for automotive, where robustness is a key characteristic, meaning that it really has to be an all-in-one design rather than stacked boards and wires.

But still, having some common hardware blocks at least eliminates the duplication in software.

If you are suggesting that i2s mezzanine be based directly on my schematics, I think that would be a very suitable approach. I don't have time to do a board right now, but I can certainly offer schematics.

mwelling, 2018-07-06:
Well our doors are open for a pull request. If you rather just point us to the schematics and have us pull the IP into this project that is also fine.

What PCB package was used?
Are the design files freely available?

Yang-96Boards, 2018-07-06:
> 
> Well our doors are open for a pull request. If you rather just point us to the schematics and have us pull the IP into this project that is also fine.

Ditto - this sounds a well fit design

Would you also consider putting microphone array on board? That would be an interesting kit Amazon Alexa team is looking for.

mwelling, 2018-07-06:
I think a microphone array might need a bit different underlying hardware.

Perhaps this would better for a third board specifically for speech recognition.

mwelling, 2018-07-06:
@Yang-96Boards do you think we could get support from Amazon for an Alexa mezzanine?

ASerbinski, 2018-07-06:
Unless you feel like blending multiple audio streams together, a microphone array would not work well with a pcm1865. While it can technically address up to 12 microphones (8 analog + 4 digital), it can only output 4 channels, either over a single dout in TDM mode (which disables the second dout), or 2 channels on each dout in i2s mode.

I'll submit a pull request with schematics, but I won't be able to today, it will take me a little while to prep them. I work in gEDA.

mwelling, 2018-07-06:
@ric96 can you add a new folder for gEDA projects.

Mani-Sadhasivam, 2018-07-07:
@Yang-96Boards We talked about this microphone array feature before but we settled with a mezzanine which just acts as a general purpose soundcard.

As @mwelling pointed out, if we get support from Amazon then we may do a separate mezzanine for their usecase. I guess this is also what Cademce is looking for!

Yang-96Boards, 2018-07-07:
> @Yang-96Boards <https://github.com/Yang-96Boards> do you think we could
> get support from Amazon for an Alexa mezzanine?

I have just sent a proposal to Amazon Alexa team - let's see what we hear
back. Will keep this updated.

ric96, 2018-07-07:
@mwelling I can create a new folder but it should go into the "other" folder since it's the only one using gEDA.

mwelling, 2018-07-08:
I think gEDA being open aligns well with this repository. Just got to remember how to use it to review the design for the pull request. :)

closed 2019-11-25 by ric96