96boards/mezzanine-community #77: LVDS -> MIPI CSI

sdrobertw, 2018-10-03:
Any thoughts on this one? 

@mwelling @ric96

mwelling, 2018-10-03:
So LVDS camera/display output converted to CSI before going to the high speed header?

Might be able to leverage something like this:
http://www.analog.com/media/en/technical-documentation/data-sheets/ADV7782.pdf

What are you trying to accomplish?

sdrobertw, 2018-10-03:
@mwelling Yeah, Joyce (96Boards PM) shared some Lontium and Lattice chips that would do the trick too, I guess...

I don't know exact reasons, I will look into it. This mezzanine idea was brought up by Yang. Thought I would at least throw it up here as a place holder in case there are any discussions around it.

Yang-96Boards, 2018-10-03:
This is to take 4 x 1080p GMSL or LVDS cameras mux into 1 CSI2 - automotive vision subsystem use case.

mwelling, 2018-10-03:
If there is a use case then I think it is certainly worthy. Thanks for posting.

Yang-96Boards, 2018-10-04:
Thanks Michael - if anyone got one of those I ll personally be buying a
dozen right away :-P... we really need this.