96boards/mezzanine-community #78: Meeting: Agenda 1st Oct 2018

ric96, 2018-10-15:
Meeting time: 4:00PM UTC
Please comment on the next meeting's agenda items:

@mwelling
@ric96
@sdrobertw
@Mani-Sadhasivam
@tbewley
@ASerbinski

ric96, 2018-10-15:
Agenda:
- Staging Folder: https://github.com/ric96/mezzanine-community/commit/079096561a69609c1e240ba9791507180c729ba0
- LVDS -> MIPI CSI: https://github.com/96boards/mezzanine-community/issues/77

ric96, 2018-10-15:
Micheal:
- Moving Audio Mezzi to crowdfund (Adam?)
  - Micheal and Adam to start talking to Groupgets to get this going.
- Contest with Hackster.io, meeting at ArmTechCom
- FPGA Mezzi using Lattice

Mani & Sahaj
- Find a decent PCB house locally in India to source sample designs to be tested before launch...

ASerbinski, 2018-10-15:
Regarding the bug in the pcm1865 driver;
I thought about fixing it, but apparently didn't actually get around to it, rather just disabled apm as a quick hack.
https://gitlab.com/HiKey960-Car/android_kernel_linaro_hikey/commit/0c2084accad551f4f23785e1db52118c894f1876

Fixing it should just be a matter of setting the winvert parameter (the last parameter) to 1. I'll try to run a build this week to confirm and then see about upstream.

mwelling, 2018-10-15:
@ASerbinski thanks for all of your effort so far.

closed 2019-11-25 by ric96