96boards/mezzanine-community #97: E-ink event Badge

ric96, 2019-05-23:
[image]
Excuse my crude drawing.

So here is the Idea. I am planing on making a badge can will be provided to EVERY attendee in a conference. So this means to keep the cost down, no active components on the badge itself.
Instead, attendees will be required to plug in their badges to a console at the registration desk and modify the badge if they so required.

Figure 1.: A 2.9 inch E-ink Display module. Same as the one provided by waveshare that can directly be controlled over SPI.

Figure 2. A Component-less PCB on which this E-ink module will be stuck. All the PCB does is to convert the FPC on the e-ink display to a On-PCB board to board connector, kind of like the one that's on a micro-bit or PCI cards. This is done to strengthen up the e-ink display and preventing people from interacting with the FPC connector.

Figure 3: This is the actual mezzanine, this has all the active and passive components like level shifters etc. This sits on a 96boards CE board on the registration desk and the PCB in Fig.2 plugs into this. then people can doodle images on their badge. Now, I do plan on adding a FPC connector as well apart from the Board-to-Board interconnect so that this mezzanine can be utilized as a generic E-Ink display mezzanine for 96Boards.

This setup helps us to reduce the cost and instead of having to make a few hundred PCBs with active and passive components we can get away with just a couple.

ukBaz, 2019-05-23:
My assumption is that if anything more than a paper/card badge is given away at a conference it is because it will be readily hackable and so usable after the conference is finished. Is that going to be the case with the configuration above? If the attendee only gets parts 1 & 2 then there will be quite a high threshold to being able to hack it after the event. Also, with it being SPI based, and SPI  not beingenabled by default on the 96Boards, will this again raise the threshold for how easy it is to hack?
My concern is about how many of these E-ink displays that will be used after the event and whether that is an environmentally responsible thing to do. If only a tiny percentage get used afterwards, then I think this looks expensive compared to paper/card badges.

mwelling, 2019-05-23:
E-paper badges with electronic behind them tend to have a life after the conference.
https://hackaday.io/project/112222-2018-open-hardware-summit-badge

If you need to buy an adapter to program the reuse dies off quickly.

If you intend on reusing them for multiple conferences then I could see it having less environment impact.