96boards/mezzanine-community #105: Serial Mezzanine v2

ric96, 2019-07-22:
Idea by @edolnx to create a more BMC-like serial mezzanine.

Here is the not very good block diagram, real PR with schematics from KiCAD coming soon:
https://docs.google.com/presentation/d/1QWmrCeDlf2ppB9VF9HgMG6vnkkTAB0eYvpYOhqfH8O4/edit?usp=sharing

Thore-Krug, 2020-10-29:
The CP2108 IC has I2C which is the Interface used by the INA260 Current sensor, but i dont know how difficult it would be to implement. But it would save on space, Hardware-complexity and Bom Cost.

Thore-Krug, 2020-11-02:
The Datasheet of the CP2108 is conflicting about the Io Voltage.  There was a Discussion about the that on the silicon Labs as well: https://www.silabs.com/community/interface/forum.topic.html/can_vio_of_cp2108be-KHYX 

Which states that the CP2018 dos NOT Support 1.8V Voltage.  so it might be necessary to choose another IC. I wonder if there are people who used the IC out of spec with 1.8V. 

Ping: @mwelling @edolnx

mwelling, 2020-11-02:
CP2103 maybe?

mwelling, 2020-11-02:
Looks like CP2103 is NRND. How about CP2105?

https://www.silabs.com/documents/public/data-sheets/CP2105.pdf

mwelling, 2020-11-02:
Wait we need 3 UARTs looking at the block diagram. Hmmmm.

mwelling, 2020-11-02:
USB replace the attiny with a USB capable microcontroller and pass UART through?

Thore-Krug, 2020-11-02:
Or a level shifter IC ? Like on the I2S Mezzanine ?!

mwelling, 2020-11-02:
That could work.