96boards/mezzanine-community #112: Type-C Portability Mezzanie

ric96, 2020-03-30:
The idea is to crate a mezzanine that will allow us to power and control a CE board via a host laptop/computer using a single USB Type-C Cable, this include power as well.

Design Considerations:
1) Power: Provided that the host system has a PD capable type-c port, we can provide 9-15v to the SYS_DCIN pins on the low speed header
2) UART: A simple usb to uart FTDI chip should suffice
3) Networking: Will a usb to usb networking be possible using the HS usb pins?
4)  a usb hub chip to have both uart and networking on the same usb type-c port.

EstevesDouglas, 2020-03-30:
Do you have part number Type-C?

ric96, 2020-03-30:
http://www.ti.com/lit/ds/symlink/tps65987d.pdf
@EstevesDouglas

Thore-Krug, 2020-03-30:
Another USB PD IC 

The Question here is if a good Supply-chain can be established for this Part. 
[IP2721.pdf](https://github.com/96boards/mezzanine-community/files/4404568/IP2721.pdf)

Thore-Krug, 2020-10-29:
LCSC is now selling the IP2721 PD IC.